SK21/AOG_RC #2: Restructure repository

mirh, 2021-01-30:
Can you purge all the binaries from git, and just attach them cleanly and neatly in the release section (bundled together or separately)?
Putting aside they make sources quite heavier, even if you half-way understand their working it's a mess to go back and forth between their folders.

SK21, 2021-01-31:
I can see where it is 'heavy'. I just downloaded the zip file. It is over 100 MB. So what your saying is to create a release and move the supporting files to there. Then just the updated install can be downloaded. The supporting files can be downloaded separately if needed.

mirh, 2021-01-31:
And move the binaries to the release section, yes. 
I'd also recommend to bring in the [.gitignore](https://github.com/farmerbriantee/AgOpenGPS/blob/master/.gitignore) file from aog to avoid "forgetting" stuff when pushing. 

Btw is "100" and "200" supposed to be a version number, or what?

SK21, 2021-02-02:
Yes a version number.

marshalab, 2021-02-03:
Sorry for the stupid question. I'm a newbie trying to figure out a lot of forks AgOpenGPS.
And where are the device diagrams for your development? Or is iron described elsewhere?

thanks

SK21, 2021-02-05:
Any of the pcbs will work with RateController. The 10 pin network header on RC4 needs more 3.3V current than the nano can provide. The nano clone here might work. https://www.seeedstudio.com/Seeeduino-Nano-p-4111.html A better option may be to use a Nano ethernet shield if you are connecting that way.

mirh, 2021-02-05:
Cool. 
Would it be possible to get some logical schematics like [this](https://github.com/Coffeetrac/AG_Rate/blob/master/AGRate_Schematics.jpg) or [this](https://aws1.discourse-cdn.com/free1/uploads/agopengps/original/1X/199a6c959acedecaec6603bb8612be2a2adf76c6.jpeg)?

I'm not sure what I need in addition to arduinos, a flowmeter and the controllable valve/PID controller

SK21, 2021-02-06:
Here are some schematics.
[RC schematic.pdf](https://github.com/SK21/AOG/files/5937063/RC.schematic.pdf)
[schematic.pdf](https://github.com/SK21/AOG/files/5937065/schematic.pdf)
[AOG schematic.pdf](https://github.com/SK21/AOG/files/5937066/AOG.schematic.pdf)

mirh, 2021-02-06:
Thanks, that was really helpful! (even though still not the same cakewalk of the normal dokuwiki instructions for autosteering)
I only need the motor driver then now. 

Why is there a capacitor on the flow input trace though? My plan was just to connect the spare components in a fancy box without the pcb.

SK21, 2021-02-06:
On one tractor I found a lot of noise on the signal. The capacitor seemed to clean it up.

mirh, 2021-03-09:
Uhm, well, now the repo is quite cleaner. Thanks. 
Is there any way to control a "master" machine switch though?
It seems wasteful to turn off all sections independently when you get at the end of every line.

closed 2021-03-09 by mirh