SK21/AOG_RC #3: RC4 PCB and parts lists?

ikidd, 2021-03-07:
I think I've looked through everything, but can't find the PCB sections and the parts lists?  Am I just blind?  In another copy of the repo I have a top level folder called RateController100 with them inside, but I was checking if there were any changes to what I have and can't find anything in the current version of the repo.

SK21, 2021-03-07:
You can find it under releases, version 1, other files. There is no changes to rc4.

joshwilsonfarms, 2021-04-01:
[image]
Is there anyway to incorporate rate control in to my section control system without using the PCB you have listed?

SK21, 2021-04-16:
You would just have to change the pin assignments to match your setup.

closed 2021-05-03 by SK21