SK21/AOG_RC #4: Use without I/O extender?

mirh, 2021-04-15:
I have a nano, and a board with 8 DPDT relays (and I should only need 5 of them). 
Is it possible you could add a switch case or alias to the mcp functions, to use the normal pinMode/digitalWrite optionally?

(btw, for maximum noobproofness, I guess you could even add a comment on line 80 of RCarduino.ino to point out that in case of error users should download the right library from the manager in the arduino ide tools section)

SK21, 2021-04-16:
I can add code to use RCarduino without a I/O extender.

mirh, 2021-04-16:
Cool, thanks. 
I have more questions than answers though now. 

Why did you switch to the analog pins?
And why only 4?

SK21, 2021-04-16:
The analog pins are the same as digital pins. If someone is using a ethernet shield those pins are not being used. The ethernet shield uses pins 7,8,10,11,12,13. You can change it to whichever pins are free.

SK21, 2021-04-16:
Pins A6 and A7 are input only and won't work for relays.

mirh, 2021-04-16:
Right, I see the comment. 
Though now, something else is escaping me. 

With your vanilla code, I'm getting relays at D2, D3, D4 and D6 to turn on, and I cannot seem to do anything else with the thing. 
If I adjust the code for Relay# to go from 2 to 9, I only get D4 and D6 to light up, and again I cannot seem to toggle them. 

Once put aside my lack of control, the FlowDir array seems to be the deciding factor... But I'm not sure how I'd be supposed to control n sections when the code (even for the MCP23017 case) has just two pins there. 
Sorry if I'm making overly dumb questions.

SK21, 2021-04-17:
For  FlowPin, FlowDir, and FlowPWM, the first element of each array is for one product. The next element of the array is for another product. For the first (or only) product the first element of the FlowPin is connected to an interrupt and senses the flow. The FlowDir and the FlowPWM are connected to the motor controller to open or close the control valve.

How are you turning on the relays? Are you using the RateController app? Might be a setting in that.

mirh, 2021-04-17:
Duh, right. I hadn't completely configured RateController, and it wasn't working. Or maybe it was just that (*understandably*) I had to \*move\* the tractor in the simulator for the section switches to turn on at all. 
I still have a couple of remarks though: 
* it should be pointed out that AgIO has to be kept open for RC to hook it (it may be obvious when it's AOG to handle the machine port, but given here the serial port doesn't have to be hooked there, it's confusing)
* I can [half-way](https://github.com/SK21/AOG/blob/2.1.3/RC%20Source/RateController/FormStart.cs#L209-L218) guess what that salmonish "SimColor" hue for *Mod* means, but if I were out there in the field I could hardly see that shade (let alone figure out if it's just an innocent warning, or something I should quite be worried about)
* just to check, is it normal switching to "fast close valve" has D5 and D6 always turned on, while switching to "motor" has only D6?

SK21, 2021-04-17:
AOG only connects to AGIO, so yes AGIO needs to open. But then to use AOG, AGIO needs to be open anyway.

In the field the simulator should be off, so Mod with be either red or green. Also in the product detail the heading shows if it is in simulation mode.

If D5 and D6 are for relays, changing the valve type shouldn't affect whether the relays turn on. Make sure D5 and D6 aren't assigned to anything else such as PWM or DIR.

mirh, 2021-04-17:
Makes all sense now, after seeing the SensorCount loops I think I got the point of the arrays arranged this way. 
Thanks very much.

closed 2021-04-17 by mirh

mirh, 2021-04-20:
Ok in retrospect I don't think I have actually understood them. 

I only have one sensor. 
But I can't only write a single pin for flow and pwm in the arrays, because SensorCount is 2. 
And if I lower it to 1, then hell break loose across many files. 

I can easily see the 0s and 1s with arduino's [InputPullupSerial](https://www.arduino.cc/en/Tutorial/BuiltInExamples/InputPullupSerial) example on D3, but I cannot get ratecontroller to understand it.