SK21/AOG_RC #5: RateController crash since 2.1.1

mirh, 2021-04-19:
Event log says we have a `System.NullReferenceException in RateController.Program.Main()`
After [checking](https://github.com/SK21/AOG/blob/2.1.3/RC%20Source/RateController/Program.cs), it seems like you crash if `HKCU\SOFTWARE\AgOpenGPS` is not present.

SK21, 2021-04-20:
Thanks, I will move the Close() statement into the If statement.

mirh, 2021-04-22:
Thanks, it works now. 
I wish people on the main program repo were this outgoing...

p.s. you probably are better off doing some git revert/rebase magic and delete all the binaries you had slip into the source code. I guess you should add a [.gitignore](https://github.com/github/gitignore/blob/master/VisualStudio.gitignore) file

closed 2021-04-22 by mirh