SK21/AOG_RC #6: Serial monitor is flat

mirh, 2021-04-28:
So.. after much tinkering (somehow, I had completely missed we are working on baud 38400) I got to read your program data inside the arduino IDE. 
The problem is I'm only getting `127,101,0,0,0,0,0,0,0,0` or `127,101,1,0,0,0,0,0,0,0` as returned values, forever. 
Do I need to use RateController to have actual feedback back?

And even then, am I supposed to do anything else besides setting simulation to off? 
I tried to connect my flow meter to either of the FlowPins (that I checked to get the correct signal with a simple code sample), but no matter what I do, calibration never senses a single count.

SK21, 2021-04-28:
You need RateController for the ino to work. Set simulation off. If you are connecting with serial use one of the connections on the comm page. The green light should come on. 
[image]

Also the Mod indicator should turn green. 

On the Products/Options for the product you are using the Module ID and Sensor ID should be 0. The indicator should be green.
[image]

The Products/Diagnostics tab will show any flow signals as UPM Applied. The tractor will have to be moving and sections turned on. Make sure the sections widths are entered again in RateController even if they are already entered in AOG. Make sure on the tab Products/Options the sections you are using have a check mark.

On the Products/Rate tab the Sensor Counts/Unit needs to be entered. Whether you use VCN or PID press the Load Defaults button.

mirh, 2021-04-29:
:\
Everything I had to do was opening AOG, not even the tractor moving, and it works now. 

...

I get why a lot of things requires the main program open, but why is that also "hard blocking" of UPM applied and sensor counts?

SK21, 2021-05-01:
Yes that could be done as well.

SK21, 2021-05-02:
Have you tried receiving the pulse counts without AOG?

mirh, 2021-05-03:
Yes, that's what I was banging my head against in the first post.

SK21, 2021-05-03:
See if it works now.

mirh, 2021-05-15:
Perfect now, thanks!

closed 2021-05-15 by mirh