SK21/AOG_RC #7: Relay control over serial?

mirh, 2021-05-01:
Sigh, I can't understand if I'm always here complaining because I'm one of the few people that got at least half-way of training in both AG and ICT, or if I'm just a bane of intelligence. 

How are you supposed to control relays from AOG, if RC takes full ownership of it?
I mean, I see the code to control them is there, but nothing triggers when I click on the various sections.  

Or is this only a limitation I brought upon me when I decided USB/serial was good enough for me?

SK21, 2021-05-02:
Serial and UDP work the same. If you mean controlling the sections with the on-screen buttons of AOG, then yes these should work. The sections should be able to be controlled from either the on-screen buttons or from remote physical switches. If you have physical switches that are turned off, these take precedence to make sure sections stay off. If you don't have physical switches then it is set to auto mode where the on-screen buttons should work.

SK21, 2021-05-02:
The on-screen section buttons previously only worked in RC auto mode. They now work in manual mode as well.

mirh, 2021-05-03:
***Well***
I'm glad you found a potential bug (I didn't notice it)
But it turns out I had just forgot I had stolen the +5V/GND cables from the relays to power the flowmeter. 😓

closed 2021-05-03 by mirh