SK21/AOG_RC #8: AOG_RC prescription maps

mappaturakiwiverona, 2021-05-04:
Hello,
I'm looking into your great app for Rate controller.

I have a couple of questions:
- Is the app ready to use variable rate prescriptions? [If yes where do I need to import them?]
- Do you have experience controlling rate on a Rauch Quantron-E display?
Thank you for supporting,
looking forward hering from you soon
Best Regards
Davide M.

SK21, 2021-05-04:
VR has been discussed. RC has no info on location in a field.  As it is now AOG sends the rate requirement (on or off) to RC based on location in the field. AOG would have to support VR prescriptions so that based on location in the field it would send for example off/rate1/rate2/rate3. These rates could be defined in RC and RC could then be modified to do VR.

I have no experience with Rauch Quantron-E display.

SK21, 2021-05-10:
I don't know when vr will be included in aog. RC could possibly support this protocol. Not enough time to work on it right now.

closed 2021-05-30 by SK21