SK21/AOG_RC #9: Some hardship on adding more relays

mirh, 2021-07-30:
Soo.. As usual I'm with my Nano, connected with a simple usb cable. 
Everything was fine and dandy, until I tried to go past the 4 relays that the "Nano pins" codepath defaults to. 
I uncommented the pinMode for Relay5 to Relay8, then same for their defines, and last but not least I set them to follow the natural progression from A4 to A7
But for some reason I can still only control the first four usual relays (and yes, I did set 8 sections in both RC and AOG)

---
p.s. I also found great.. puzzling when it came to interpreting this line. 
https://github.com/SK21/AOG_RC/blob/f1fbfcb5fb170020fd8f94c0bf1c1fda9da2faf9/RCarduino/RCarduino.ino#L28
Could it be named to something like *UsePhysicalSwitches*?  Also, the comment could maybe be changed to "1 to use commands from hardware switchboard"

SK21, 2021-07-30:
[image]

Did you set the switch # for each of the other sections? Do you have 8 switches? If not one of the existing switches that you have can control some of the new sections.

I don't think A6 and A7 will work. They are input only on a Nano.

Line 28 attempts to clarify whether the switches used to control the sections are connected (local) to the rate PCB or are the switches on a separate pcb dedicated to switches or the sections are controlled on-screen. 0 means no switches connected to the rate pcb. 1 means switches are connected to the rate pcb.

SK21, 2021-07-30:
[image]

Also need to uncomment lines 35 to 38 of the Relay.ino

mirh, 2021-07-31:
Perfect, that was it!
(I wonder if it couldn't be made a bit more obvious? perhaps already just with a comment)

> Line 28 attempts to clarify whether the switches used to control the sections are connected (local) to the rate PCB or are the switches on a separate pcb dedicated to switches or the sections are controlled on-screen.

Yes, I had eventually imagined you were aiming for something like [this](https://www.youtube.com/watch?v=kI321iBwe4U) or [this](https://www.youtube.com/watch?v=K2U5_X6FHQg). 
It's just.. I don't know, that wording isn't the clearest label on the shed.

closed 2021-07-31 by mirh