SK21/AOG_RC #10: Proportional only control?

mirh, 2021-08-18:
Long story short, my flow valve has a motor that works with polarity reversal. 
And through the power of DPDT switches, we could just say it has two commands/states: "more" and "less". 

I tried to half ass together something in *Motor.ino*, but clearly there's so much more to it. 
Would it be too hard and disruptive to add support for this?

SK21, 2021-08-18:
Are you using the motor option or the standard valve?

mirh, 2021-08-18:
I tried to tinker with the standard valve code, after having retooled the PWM pin to be just a digital one outputting the opposite of the direction state... but it kinda failed. 
But I think this would be worth a new type of motor tbh (is "proportional" even the right word? I have no control on the velocity, maybe "linear valve" would be better?) since nothing in the PID settings tab would apply.

SK21, 2021-08-19:
I am not clear on how your motor works. A standard valve has more and less commands. The motor control varies the voltage from 0 to 12 v. 

Do you have a model # or link to a page about the motor?

mirh, 2021-08-19:
I have [this brochure](http://www.bertolinipumps.com/irrorazione_accessori/cp_mot.pdf#page=2) (last one) about the body, and [this](http://www.bertolinipumps.com/prodotti/cataloghi/catalogo-multiduplospray.pdf#page=2) should be the valve-only. 

> A standard valve has more and less commands.

Yeah, but you also have PWM to regulate speed, while "effect direction" is the state of a single relay. 
Here instead, you just supply +12V or -12V (which at the end of the day I made to correspond to two separate relays) and that's about it.

SK21, 2021-08-21:
In the last brochure about the valve it looks like the motor only controls on/off, not flow.

[image]

mirh, 2021-08-21:
It's on/off in a sense, yes. 
But the "obturator" can end up sitting in more than just two states, if I can explain. 
And this way (since polarity reversal makes for *two* separate "ON" eventually) you can control flow.

bduncan175, 2021-09-17:
It sounds like his valve is similar to a standard raven flow control valve that is plumbed as a bypass system, but can also completely shut off. https://www.spraysmarter.com/raven-precision-1-poly-control-valve-flanged.html?gclid=CjwKCAjw-ZCKBhBkEiwAM4qfF1c-M0oKpccFcGTAFIvWbRJOzAW5RACnZJ9z2HZ7LF6Ss77eIKoxUhoCoMEQAvD_BwE

mirh, 2021-09-17:
Sorry, I snafued the valve brochure. [This](http://www.bertolinipumps.com/prodotti/cataloghi/catalogo-serieviking.pdf#page=1) should be the right one. 
To reiterate again: with +12V (i.e. one relay) the ball rotates in one direction, with -12V (an other relay) the opposite one.

SK21, 2021-10-04:
So you would like to control the pressure with 2 relays, one for increase and one for decrease? A motor controller would not be used. I could change the code to add another control type that would use relays instead of a motor controller.

Motor.ino:
[image]

SK21, 2021-10-04:
Should be:

[image]

SK21, 2021-10-05:
It could also work with a motor controller if the PWM is set to maximum. This would give 12V.

[image]

mirh, 2021-10-07:
> A motor controller would not be used.

I think the confusion is due to the valve mechanism technically having one, but the DPDT wiring I used making it logically be just two switches (I mean, I'm not sure how else I could have controlled polarity inversion otherwise). 

***Anyhow***
```C
// use 2 relays for flow control
if (FlowEnabled[i])
          {
              if (pwmSetting[i] > 0)
              {
                  // increase
                  digitalWrite(FlowDir[i], 1);
                  digitalWrite(FlowPWM[i], 0);
              }
              else if (pwmSetting[i] < 0)
              {
                  // decrease
                  digitalWrite(FlowDir[i], 0);
                  digitalWrite(FlowPWM[i], 1);
              }
              else
              {
                  // no change
                  digitalWrite(FlowDir[i], 0);
                  digitalWrite(FlowPWM[i], 0);
              }
          }
          break;
```
was exactly what I needed (at least when I replaced the fast close valve block for testing purposes). 
The only thing is.. since this has no derivative control, at least when near the target rate shouldn't you have some kind of margin of tolerance to not flip-flop forever the relays?

mirh, 2021-11-01:
Thank you so much!
With combo close now everything is just right (I still have some idea for target rate adjustments on-screen buttons, and hooking the "master valve", but that will be for another time I guess)

Do you accept donations by any chance?

mirh, 2023-04-12:
Ok well, I quite literally just re-repeated the code above and it worked this time. 
If the merit wasn't to be with some of your latest commits, I suppose it could be with me possibly having mixed up the pin numbers due to the different order the original lines have. 
https://github.com/SK21/AOG_RC/blob/feeb3e6a7ddbb1c69c3beb8bc7b5be15e5b7a55b/Modules/Nano/RCnano/Motor.ino#L24-L32
The only "novelty" is that I also have to remember to set _PWM maximum_ to 255.

mirh, 2023-04-29:
I don't know if this couldn't be an offshoot of #30, but as if it wasn't enough I cannot get the Every Nano to respect that
```
  // no change
  digitalWrite(Sensor[i].DirPin, 0);
  digitalWrite(Sensor[i].PWMPin, 0);
``` 
bloody code. 
Once the pin turns on once, it doesn't shut up until I disconnect the communications (or well, I enable auto mode *\_-\_*).

mirh, 2025-04-27:
I'm once again in need of a hand with this more/less valve.
Ever since 2.3.6~2.3.7 (I'm gonna guess due to the different way the deadband and other calculations were changed?) the relay that should increase the flow rate keeps staying on even when the implement is turned off.
And starting from 3741da803d9b6d49f46cb74d0c5ecd4aff446a49 it's instead just dead.