SK21/AOG_RC #12: Master valve support

mirh, 2022-02-02:
I didn't really want to bother you with yet another "personal customization" request, but I just noticed `UseSwitchedPowerPin` seems kinda half-way it already. 
You would only need to move it from "per whenever the system boots up" to "per actual usage" (i.e. whenever there is at least a section turned on)
This would be particularly helpful wear-wise, so when you reach the end of the crops row or when you manually toggle all sections, you don't have to "strain" n-th relays and valves but just a single one.

SK21, 2022-02-03:
As it was originally intended this change would not work correctly. This could be implemented on a per user basis. The code could be modified to fit your needs.

mirh, 2022-02-03:
What was the original purpose then? Something other than pulling open some master switch?

SK21, 2022-02-04:
It was to provide power for the rate control valve.

mirh, 2022-02-04:
Oh, I see, as opposed to a valve where control and power are on the same line. Sorry. 
Well then, my feature request still stands. 

Whenever work is being done you would have such relay enabled, and when an "action" would entail that no section is powered anymore you just leave them alone, and instead toggle this.

SK21, 2022-04-09:
In the Dev branch the Rate App will now do Master relay. On when sections are on, off when all sections are off and the section relay aren't moved.

mirh, 2022-05-08:
Sorry for the delay.. lots of bad stuff going on. 
But I'm quite lost with the new rewrite of the arduino sketches (I can sense perhaps you wanted to simplify/unify the MCP/nano-only cases, but comments seem skewed just for the former now). 
Where would I even specify the "master relay"? I don't see any such entry in the PCBpinConfig struct, nor I could spot any code designed for this functionality. 

p.s. as I already said plenty of times, please add [some](https://github.com/github/gitignore/blob/main/KiCad.gitignore) [gitignore](https://github.com/github/gitignore/blob/main/VisualStudio.gitignore) to the project. The repository is just so heavy.

SK21, 2022-05-14:
Master relay is set in the rate app under the menu option relays. On the relays page you choose the type of relay. Section, slave, master, power or Invert_Section.

mirh, 2022-09-20:
Ok sorry again.. When I found new time for tinkering, I eventually had instead to debug some big annoyance with AOG itself. 
What to say then? 

Your PCBsetup app is pretty nice, but I have been banging my head for a day about what I should do for RC to work. 
Should I upload with the arduino IDE the sketches in the modules folder as usual? Is PCBsetup supposed to take care of that too? Is the whole design even supposed to work with just a bare Nano?
Because no matter what I tried, the `Mod` indicator doesn't light up. 

I know something ***is*** happening because every time I connect or disconnect from the Comm window, I can hear the relays clicking.. But red colour doesn't seem good. 
Btw the monitor in your test app gets stuck at "tarting I/O Expander" (sic.)

SK21, 2022-09-22:
Is it the timing in AOG that you had to debug?

PCBsetup can load the firmware and the settings. It is meant to be used instead of an IDE, so no knowledge of programming is required. The IDE can be used if preferable. 

I just erased a Nano with a blank sketch. Then I used PCBsetup to upload the Nano rate firmware. It showed fine in the serial monitor.
[image]

I connected the Rate App to the Nano comm port, ran AOG and started a field. Mod turned blue and then green. Blue indicates the module is sending but not receiving.

[image]

Have you tried uploading the Nano rate firmware? You may need to switch bootloaders.

[image]

mirh, 2022-09-22:
> Is it the timing in AOG that you had to debug?

https://github.com/farmerbriantee/AgOpenGPS/issues/264

> Have you tried uploading the Nano rate firmware? You may need to switch bootloaders.

I also did that. 
*Modules* with the IDE, the default from pcbsetup, and even the latest .ino.hex from *Resources*. 

It doesn't change anything. 
In serial monitor, I only see `RC Nano : 17-Jul-2022` (every time) and then once every n attempt, there's some random part of the string "Starting I/O Expander". 

Btw when I tried "use old bootloader" it almost seemed like I had bricked the arduino, and I could only recover from the IDE.

SK21, 2022-09-22:
I am not sure what else you can do. Does uploading the "Blink" sketch from the IDE examples work?

mirh, 2022-09-22:
:\\
No? I don't have any wemos (or pcb for all that it matters). 
Where is that even mentioned? And why would ethernet be required even if the corresponding button isn't selected?

SK21, 2022-09-22:
The wemos on the pcb connects to Nano rx/tx. These pins are also used by the usb connection. The ethernet shield doesn't use rx/tx so there is no conflict. 

If you don't use the wemos then there can be no comm conflict.

mirh, 2022-09-23:
"Comm" is just fine indeed (be it in the namesake RC window, or the two green lights in pcbsetup). 
I don't have any wi-fi or ethernet, I'm not getting conflicts. 
I just have literally a Nano and a straight usb cable to pc. Is this not supposed to work anymore?

SK21, 2022-09-23:
Yes usb will work. I just tried it in the previous posts. Just a nano and a usb cable.

closed 2023-02-26 by SK21

mirh, 2025-04-27:
Since 3.7.9, when I stop the sections (that is, all of them together like with the button in AOG) rather than just the master relay turning off, every single one of them switches.