SK21/AOG_RC #21: Puzzling relay situation

mirh, 2023-02-18:
So, long story short (hoping this is not related to #15?) I have this Nano with relays connected to the analog pins. 
Since PCBSetup expects the digital pins codes, I set Relay1 to Relay5 the numbers from 14 to 18. 

...
This doesn't seem to do anything no matter what I did in RC. 
After desperately adding 16 sections to my virtual sprayer in AOG, I figured that the physical relays finally came to life when I went to control section 9 to 13 (which under the default RC settings are assigned to Relay9 to Relay13). 
And for the life of me I cannot understand what's happening (putting aside that for the moment I'm not even caring for the flowmeter). 

Could it be that [last year](https://github.com/SK21/AOG_RC/commit/a905b8e1699570078a962c9ac47ed35eb792e4f3#diff-8a84c8f70f782e5151250551821e26f556eee1f2c69baefcc0468aa41a62bd88L80) you dropped the vanilla Nano pins assignment table?

SK21, 2023-02-18:
There was an error in the relay code. It is fixed now.

SK21, 2023-02-26:
Can you give me an example of the glitchy buttons.

mirh, 2023-02-26:
Their images are missing. You can still use some of the settings, but opening products will crash the application. 
Again, this should be easily reproducible if you just compile release (I was targetting VS 2022 and .NET 4.8 but I don't think it matters).

SK21, 2023-02-26:
I am using vs 2022 and .net 4.7.2.   Again I can't reproduce your results. No images missing, no crashes.

mirh, 2023-02-27:
Well.. I'm not able to reproduce it either. 
I guess it's still a bit funny that I can compile it even though I did uninstall 4.7.2 (and I didn't retarget anything) but whatever.

closed 2023-02-27 by mirh