SK21/AOG_RC #22: RC is dead when base rate is astronomical

mirh, 2023-03-19:
Because of reasons (likely related to my going back and forth between new and old versions, but idk really) my base rate just so happened to go out of bounds. Like in the hundreds of thousands or millions. 
Nothing crazy or noteworthy by itself, except for the part where (until I reset it back) no signal seemed to be sent out to my relays. 

Of course wrong inputs could only lead to wrong behaviour, but "nothing happens" seems relatively insidious to track down to this. 
I suppose you could add extra drippy validation in the *Product* page too, but the proper fix should probably go towards whatever glitchy code that parses/handles `lgBaseRate` further downstream. 

p.s. in somewhat related news, where did *deadband* go after 2.3.7?

SK21, 2023-03-19:
I have updated the code to add extra checks for rate validity. 

Deadband is fixed in code and could be edited there. 
[image]

mirh, 2023-03-19:
> I have updated the code to add extra checks for rate validity.

I was expecting whatever case of overflow getting either the app or the board stuck to get handled, but I guess I'm not the sharpest tool in the shed here. 

> Deadband is fixed in code and could be edited there.

I had a suspect about that, but why is this even the case now?
Like, if I didn't have a [reference](https://discourse.agopengps.com/uploads/default/original/2X/7/730bb55ac1352db311d1470141d659181b91da9d.jpeg) from the old UI I wouldn't even know what kind of unit its value is in.

closed 2023-03-19 by mirh

mirh, 2023-03-22:
I can confirm it seems to be switching between the new RC and the one in 2.1.13, to absolutely send values to the sky... even though which ones get affected is somewhat random for some reason.

mirh, 2024-07-22:
Happened again between 2.3.5 and 3.7.14 (and/or viceversa)