SK21/AOG_RC #26: Target rate can get out of control all by itself, crashing RC

mirh, 2023-04-12:
So.. while I was investigating #10, after yet another lackluster run I just so happened to turn off all the sections in AOG (I'm listing from the top of my head everything that I can remember of the preceding seconds, but I don't really know if it they are useful/related).
I went back to the arduino IDE (or whatever other window that it was, not like it matters with RC always being on top) when I noticed that the target rate was quickly rising. 

From around ~500-ish that I had set it, after 20-30 seconds it had skyrocketed to 40000, and counting. 
And then RC crashed with this error. 
```
************** Exception Text **************
System.ArgumentException: Must be between 0 and 50,000
   at RateController.clsProduct.set_RateSet(Double value)
   at RateController.clsSectionControl.ReadRateSwitches()
   at RateController.FormStart.timerMain_Tick(Object sender, EventArgs e)
   at System.Windows.Forms.Timer.OnTick(EventArgs e)
   at System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   at System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)
```
I don't know what settings/buttons/functions could increase the base rate unprompted, but one of them must be bugged (I was using the inos and [binaries](https://github.com/SK21/AOG_RC/tree/FixCal2/RateControllerApp) from your *FixCal2* branch FWIW).

SK21, 2023-04-13:
Were you using the on-screen rate increase switch or were you using a physical switchbox?

mirh, 2023-04-13:
No physical switchbox. 
I believe I did have previously played with the on-screen buttons then, but not at any near point in time to the aforementioned events (AFAIR the corresponding window wasn't even open).

SK21, 2023-04-13:
I have adjusted the code to handle the error without crashing. I am not sure how it could have increased by itself. Does the switchbox use a momentary switch for rate?

mirh, 2023-04-13:
I don't have anything outside of the arduino, the sections/valve relays and the flowmeter. 
I was running *RCnano* with a slightly modded `Motor.ino`, but that doesn't sound like it could influence rate...

Maybe yet [another](22) hijinx of intermingling an anciently created `Default.rcs` with the new versions? :\\
Anyway, putting aside that maybe the crashing wasn't all that bad considering the alternative was the undefined behaviour of infinite growth, I'll just close this because I don't have any other idea and after 2 years (lol) I'm very much wishing to have finally reached perfection with my setup.

mirh, 2023-05-20:
The target rate ballooning happens when you rapidly start and stop the calibration process while also playing with the rate up and down buttons.