SK21/AOG_RC #28: Lone pin turns on with every initialization

mirh, 2023-04-16:
I'm pretty much in a honeymoon phase now that everything works like in my smoothest dreams, but I have this quirk every time I power on my Nano (or I connect it to the pc, or to RC). 

I could be expecting the master valve alone to create this problem, or somehow all the section relays to trigger together because whatever code reason I wasn't educated enough to figure out... Yet, instead, only section number 5 (wired and set to D18) has this kneejerk reaction. Like, I'm doubly dumbfounded this way. What could it have of special?
(this isn't *really* much of a problem then.. but still, it's kinda ruining perfection)

mirh, 2023-05-16:
Ok well... It seems like this is some kind of bootloader voodoo, no possible fault of yours (though it's crazy that it also somewhat happens even on the nano every that should use something completely different from optiboot)

    avrdude -c arduino  -p atmega328p -b 115200 -P port -t

Whenever you enter programming mode (or you reset the board thereof I guess?), you get the issue on D18/A4. 
Worse even, I found that D19/A5 is *constantly* pulled high for the entire duration of the ordeal, not just for the blink of an eye. 
Thankfully I do have quite the number of other pins that I can employ...

mirh, 2023-05-17:
Actually nevermind, my understanding of the bootloader modes was seemingly limited. 
Whatever happens when you use the command above, if I upload an empty sketch it stops. 
So.. I do guess it's something with the code here? Even though I cannot seem to find them anywhere...

mirh, 2023-08-12:
Ok, so, I did some bisecting. 
This was introduced between 2.2.0-beta and 2.2.0-c, where even (at least) D4 was pulled up for some reason. 
From 2.2.0-d instead (and ever since, including the last git) only the usual D18 and D19 seems to be on instead.

SK21, 2023-08-13:
I don't know what is causing this behaviour. A work around would be to use a MCP23017 (RC12_PCB) or PCA9555 (RelayDriver5 in modules) to control the relays.

mirh, 2023-08-13:
I mean, my workaround was eventually just to move around the wires and set up with the same relay D6. 

Then, idk about that D4 oddity but D18/A4 and D19/A5 also happen to be the SDA and SCL pins. Does this ring a bell?
I reckon this could only have been either 6ab3da64519371f326c5a698758ea8aa0e7e9589 (pulling in some nasty new header?) or a905b8e1699570078a962c9ac47ed35eb792e4f3 reinventing too much of the wheel. Admittedly, my own job should have been already yesterday to assess this last question but after 2 hours I was tired.

mirh, 2023-08-16:
BEHOLD THE LINE (which somehow even works without the supposedly required header)
https://github.com/SK21/AOG_RC/blob/6578952a5a167dcfa5c80adb3c9b369a197101ed/Modules/Nano/RCnano/RCnano.ino#L227

closed 2024-02-19 by SK21