SK21/AOG_RC #33: Add minimum and maximum limits of flow per nozzle

mirh, 2023-07-24:
FWIW this is just a super finicky QOL feature, don't feel pressed into it and please prioritize far more important issues. 

So long story short, as you are coming to a halt at the end of a line, your speed will lower. And so, with it, rate. 
Of course there's no way for RC to know your motives for slowing down, and not like that's even much of a distinction worth making anyway (as long as you are within a field, you want to have coverage for every single meter).

***Yet*** below a certain pressure threshold, my nozzles don't have enough force to vaporize the water anymore and they just "pee". 
This also further carries over on the next row (when you are then re-starting from a standstill) where the first meters will just drip a trickle of water rather than the full ass spray. 

The "minimum cutoff speed" setting that AOG has in the *attachment sections* menu does kinda help I guess, but this use case isn't exactly its purpose and it could not account for the difference between having 2 or 10 sections running. 
So for this reason, I was envisioning a "minimum nozzle rate" option (measured in litres per second) that could clamp whatever the current rate per hectare is. Because sometimes using a bit too much liquid is better than "practically none". 

p.s. I don't personally see a scenario for the maximum limit, but nonetheless I thought it should be a cakewalk to implement once the logic for its mirror-opposite was already in place

SK21, 2023-07-25:
There is a minimum UPM setting in product options. It is not based on working width or number of sections on. It could be changed to this.

mirh, 2023-07-26:
I reckon you are right, though I don't think width should have any relevance here.

closed 2024-02-19 by SK21