SK21/AOG_RC #35: The application stops when I want to load a shapemap.

SLOWiK83, 2025-01-27:
The application stops when I want to load a shapemap. This only happens when I start AOG.

SK21, 2025-01-27:
When AOG is running RC stops or freezes? Is there any information in the error report on the help page?

SLOWiK83, 2025-01-27:
This is the error that forces the application to stop, I mean to turn it off. Unfortunately, my windows is in Polish.

[image]

Here is the text of the error:

Szczegółowe informacje na temat wywoływania debugowania w trybie JIT (just in time)
zamiast tego okna dialogowego znajdują się na końcu tego komunikatu.

************** Tekst wyjątku **************
System.IO.FileNotFoundException: Nie można załadować pliku lub zestawu 'ProjNET, Version=2.0.0.0, Culture=neutral, PublicKeyToken=a1f5bdcd9c6f5740' lub jednej z jego zależności. Nie można odnaleźć określonego pliku.
Nazwa pliku: 'ProjNET, Version=2.0.0.0, Culture=neutral, PublicKeyToken=a1f5bdcd9c6f5740'
   w RateController.Classes.MapZone.CalculateArea(Polygon polygon)
   w RateController.Classes.MapZone.Hectares() w G:\Sync\Documents\GitHub\SK21\AOG_RC\RateAppSource\RateController\Classes\MapZone.cs:wiersz 39
   w RateController.Classes.MapManager.UpdateRates() w G:\Sync\Documents\GitHub\SK21\AOG_RC\RateAppSource\RateController\Classes\MapManager.cs:wiersz 257
   w RateController.Menu.frmMenuRateMap.UpdateForm() w G:\Sync\Documents\GitHub\SK21\AOG_RC\RateAppSource\RateController\Menu\frmMenuRateMap.cs:wiersz 441
   w RateController.Menu.frmMenuRateMap.Manager_MapChanged(Object sender, EventArgs e) w G:\Sync\Documents\GitHub\SK21\AOG_RC\RateAppSource\RateController\Menu\frmMenuRateMap.cs:wiersz 276
   w RateController.Classes.MapManager.LoadMap(String path) w G:\Sync\Documents\GitHub\SK21\AOG_RC\RateAppSource\RateController\Classes\MapManager.cs:wiersz 169
   w RateController.Menu.frmMenuRateMap.btnLoad_Click(Object sender, EventArgs e) w G:\Sync\Documents\GitHub\SK21\AOG_RC\RateAppSource\RateController\Menu\frmMenuRateMap.cs:wiersz 80
   w System.Windows.Forms.Control.OnClick(EventArgs e)
   w System.Windows.Forms.Button.OnClick(EventArgs e)
   w System.Windows.Forms.Button.OnMouseUp(MouseEventArgs mevent)
   w System.Windows.Forms.Control.WmMouseUp(Message& m, MouseButtons button, Int32 clicks)
   w System.Windows.Forms.Control.WndProc(Message& m)
   w System.Windows.Forms.ButtonBase.WndProc(Message& m)
   w System.Windows.Forms.Button.WndProc(Message& m)
   w System.Windows.Forms.Control.ControlNativeWindow.OnMessage(Message& m)
   w System.Windows.Forms.Control.ControlNativeWindow.WndProc(Message& m)
   w System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

OSTRZEŻENIE: rejestrowanie informacji o powiązaniach zestawów jest wyłączone.
Aby włączyć rejestrowanie błędów powiązań zestawów, ustaw wartość rejestru [HKLM\Software\Microsoft\Fusion!EnableLog] (DWORD) równą 1.
Uwaga: z włączeniem rejestrowania błędów powiązań zestawów wiąże się pewien spadek wydajności.
Aby wyłączyć tę funkcję, usuń wartość rejestru [HKLM\Software\Microsoft\Fusion!EnableLog].

************** Zestawy załadowane **************
mscorlib
    Wersja zestawu: 4.0.0.0
    Wersja Win32: 4.8.9290.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework/v4.0.30319/mscorlib.dll
----------------------------------------
RateController
    Wersja zestawu: 1.0.0.0
    Wersja Win32: 1.0.0.0
    CodeBase: file:///C:/Users/user/Desktop/RateControllerApp%204.0.0%20B5/RateController.exe
----------------------------------------
System
    Wersja zestawu: 4.0.0.0
    Wersja Win32: 4.8.9282.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Windows.Forms
    Wersja zestawu: 4.0.0.0
    Wersja Win32: 4.8.9251.0 built by: NET481REL1LAST_C
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System.Drawing
    Wersja zestawu: 4.0.0.0
    Wersja Win32: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
System.Configuration
    Wersja zestawu: 4.0.0.0
    Wersja Win32: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll
----------------------------------------
System.Core
    Wersja zestawu: 4.0.0.0
    Wersja Win32: 4.8.9277.0 built by: NET481REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
[... 83 lines cut ...]
    Wersja Win32: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_pl_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------
mscorlib.resources
    Wersja zestawu: 4.0.0.0
    Wersja Win32: 4.8.9037.0 built by: NET481REL1
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_pl_b77a5c561934e089/mscorlib.resources.dll
----------------------------------------

************** Debugowanie w trybie JIT **************
Aby włączyć debugowanie w trybie just in time (JIT), plik .config tej
aplikacji lub komputera (machine.config) musi mieć
wartość jitDebugging ustawioną w sekcji system.windows.forms.
Aplikacja musi być również skompilowana z włączonym
debugowaniem.

Na przykład:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

Gdy debugowanie JIT jest włączone, wszystkie nieobsługiwane wyjątki
są wysyłane do debugera JIT zarejestrowanego na komputerze
i nie są obsługiwane w tym oknie dialogowym.

SK21, 2025-01-27:
What version of RC are you using? Does ProjNet.dll show up in the app folder?

SLOWiK83, 2025-01-27:
4.0.0. beta 5 from 3 days ago.

I have corrected my previous post, it is now from a computer with English language.

I have performed tests on another machine, because the first tests of AOG v6.6.2 and RC on the first computer, caused that after uploading the maps, I enjoyed the working application for a moment and driving around the field and the variable value... I can't name it, here is a photo

[image]

SK21, 2025-01-27:
That shows the size of the zone in hectares or acres.

mirh, 2025-01-28:
Do older versions work?

SLOWiK83, 2025-01-28:
Yes, the functionality of uploading map files worked fine in the previous RC version. Maybe I had an error once, but not with the "arrow up" button in the "MAP" section, but with the "import" button, I had such an exception in Windows.

SK21, 2025-01-28:
The last version added the area counter with ProjNet.dll. Did you find it in the app folder? Is this still not working?

SLOWiK83, 2025-01-28:
Here is my file list, in the latest RC version, there is no the file you are asking about.

[image]

SK21, 2025-01-28:
I just checked the release and ProjNet.dll is included. I don't know why it isn't in your app folder. Could Windows Defender be blocking the file?

mirh, 2025-01-29:
It'd be probably quicker if you could tell what was the first version that was broken.

SLOWiK83, 2025-01-29:
Very strange situation with this application. Initially I tested 4.0.0-beta.4, which was unfinished and generally did not work properly.

As soon as version 4.0.0-beta.5 came out I launched it and initially everything worked and I was happy that the problems were solved. However, at some point it stopped launching, which I described at the beginning of this thread (Issue).

Indeed! I was missing the file that SK21 talks about. I immediately downloaded RC again, but this time the application did not launch at all :/

today I looked at GitHub\AOG_RC\RateControllerApp and here's a surprise, the set of files to download is different, much smaller.

Are you sure everything is OK with the file repository on GitHub?

[image]

[image]

SK21, 2025-01-29:
The first list of files is from the Master branch. It has less files. The Development branch has more files but it was missing a couple of folders. These were blocked by gitignore. This has been fixed. If you had downloaded the source files this was a problem. The zipped app folder in the release would have still worked as the missing files were included there. Now both the source zip and the zipped app folder contain the missing files in the new release 6.

SLOWiK83, 2025-01-30:
This time(beta6) the Error is as follows, I attach a photo, and the contents of the error in the text file. It appears after uploading the Application Map file, while I click on the areas on the map, indicated by the yellow arrow. 

1. I am beginning to wonder if the Application Map file is correct, as there is one distinctive vertical line in the left part of the zones indicated by the red curves. Do you also see it?  I can even say that the yellow arrow points directly at it. 

I am also attaching a file of my maps.

I would appreciate your help and thank you for your prompt response to the problem so far.

Translated with DeepL.com (free version)

[Application Map - Shape.zip](https://github.com/user-attachments/files/18602080/Application.Map.-.Shape.zip)

[image]

[error1.txt](https://github.com/user-attachments/files/18602087/error1.txt)

2. The question is how to upload the map ? With which button ? Are the two the same ? In the picture 'button to import' I showed two buttons described A and B, to find out which is better for importing. 

[image]

SK21, 2025-01-30:
That one zone is not formed properly. The vertical line on the right side does not complete the zone. I have added error handling to prevent the crash.

To convert a shape file to match the attributes of a RC shape file use 'Import'. This allows you to cross reference attributes from the file to be imported to a new shape file that has the attributes that RC recognizes.

To load a RC shape file use the button with the up arrow. To save a RC shape file use the button with the down arrow.

There is help available for the page using the '?' button. It provides a pdf file for the page. It is in English but it should be able to be translated on-line.

[test.zip](https://github.com/user-attachments/files/18603039/test.zip)
[image]

SLOWiK83, 2025-01-31:
Thank you for your help so far. I have now uploaded the map without those vertical lines and it works fine. The topic of where these lines come from is still to be solved. The portal from which I export these maps shows me on other applications for reading such maps that there are no vertical lines there. I'm afraid that I'm generating them during the import into RCapp. 

As a next step, I need information on how to get RC to show the VR Target value, and other values, mileage, area? 

At the moment when I run AOG, the marker moves around the map in RCapp , but no values in RCapp change. Although, when I approach the map file value, the dose values are in the file assigned to each zone.

SLOWiK83, 2025-01-31:
I do the same as in the video and unfortunately only the acreage(area) changes in my case. Instead of the zone name it is Unnamed Zone, and with Product 1 it is zero all the time and nothing happens. 

Do you guys have some other version ? Is it a good idea to delete the directories from C:\nDocuments and C:\nAppData.nameLocalRateController before this operation ? 

In your video you can't see how you use the up arrow button, and without it there will be no curves with zone division on the map. 

I really want to get this functionality working, but I guess I'm out of luck :-( 

Can You tell me, what app do you use for recording desktop ?

SK21, 2025-01-31:
I am using the same version as the latest update. It is not necessary to delete directories but it shouldn't hurt to do so. The up arrow on the import form just loads the file to be imported. Once the file is saved with the new name it needs to be opened on the main form. Did you try the test.zip file further up in the discussion? If the zone name does not change or the Product A rate it would seem that these attributes were not cross-referenced correctly.

For screen recording I use this:  https://github.com/obsproject/obs-studio
Set the output format to mpeg-4 for smaller files.

SLOWiK83, 2025-02-03:
I already know everything, only thanks to your last post, I understood the essence of the matter. That first IMPORT (cross-referenced), then uploading what I saved using the IMPORT method. And uploading using the arrow button (LOAD).

Now everything works for me, thank you very much for your help and patience. It is worth including this in the help file, because it is not intuitive.

SK21, 2025-02-03:
Thanks for your feedback. Here is the revised file.

[frmMenuRateMap.pdf](https://github.com/user-attachments/files/18639374/frmMenuRateMap.pdf)

SLOWiK83, 2025-02-03:
Great, I'm glad I could contribute a little to your efforts. I have one more topic, but I'll open a new thread because this one is getting too long.