SK21/AOG_RC #36: Maps Generated after treatment

SLOWiK83, 2025-02-03:
I wanted to ask if RateControllerApp has the ability to export Post-Construction Maps? I don't know if the translator translates well.

The point is that application maps generated by FMS should generate another file in the tractor's on-board systems after the treatment, in which the actual/real parameters of the performed treatment will be saved, such as the dose and the area covered by this dose.

Is there something like that in your application, or will it be implemented?

SK21, 2025-02-03:
There is no feature like that in the app. It could possibly be implemented. Do you have an example file?

SLOWiK83, 2025-02-14:
I was able to access sample as-built maps from a specific agricultural equipment manufacturer. 

https://github.com/JohnDeere/SampleData/tree/master/Shapefiles

closed 2025-06-17 by SK21