SK21/AOG_RC #37: UART Communication

SLOWiK83, 2025-02-03:
Do I see correctly that in the latest RC version, there is no longer a module responsible for UART configuration? And that the only communication option is TCP/IP?

SK21, 2025-02-03:
Yes. The serial communication was very unreliable. Communication over serial with the switchbox would sometimes lock up the whole application.

SLOWiK83, 2025-02-03:
OK, so if I don't have a network built yet and there's nothing to receive my data, then tell me how I can monitor packets sent by RC? WireShark? But how are they generated? Is it BroadCast? Or what is the destination of such packets? Or does the connection from the module generate packet traffic?

SK21, 2025-02-03:
PGNs are sent out over udp. Here is some example code:  https://github.com/SK21/AOG_RC/tree/master/Modules/Teensy%20Rate/RCteensy

SLOWiK83, 2025-02-04:
Do you have knowledge of how such a data frame will look like what will be the source IP address, destination IP address, port ? Meaning the source address, will be the same as the IP of the computer on which RC is running, but I have trouble catching such a frame. Is some more advanced configuration needed for it to be sent during the AOG run, where the RC controls the dose ?

SK21, 2025-02-04:
Rate App comm with modules:
Receive Port 	29999
Send Port	28888
Send from Port	1480

The IP of the app is set on the modules network page. It is one of the subnets available on the PC. The subnet can be sent to the modules from there.

Rate App comm with AOG:
The App communicates with AOG over Windows loopback.
Receive Port	17777
Send Port	15555
Send from Port	1460

Modules comm with App:
Receive Port	28888
Send Port	29999
Send to IP	subnet + .255			ex 192.168.5.255
Send from IP	subnet + (.50 + module ID)	ex subnet 192.168.5, ID=1, 192.168.5.51

[Rate_PGNs.pdf](https://github.com/user-attachments/files/18657671/Rate_PGNs.pdf)

closed 2025-02-17 by SK21

mirh, 2025-04-28:
I don't want to pretend to be an expert (I'm not even using a PCB, let alone resistors or whatnot) but this is quite puzzling.
First because at least with my meagre nano and 8 relays tops, I don't remember for at least these kinds of problems (and regardless of who is the cause, if a program on the computer side freezes then it seems hard to blame anything else).

But secondly because I feel like this could close doors? Like, assuming you wanted to implement some kind of bluetooth support, the first thing you'd try was serial over it.