SK21/AOG_RC #39: Unexpected vertical lines on maps

SLOWiK83, 2025-02-17:
There is a problem, because loading maps from our system often generates vertical dashes, as in the following picture.

[image]

This is a zone generation bug that generates an error and crashes the application when a vehicle enters the zone affected by this vertical dash 

When I generated the map from the cloud application for agriculture and opened it with another application to visualize the shapes generated on the fields(dosage zones), there was no vertical dash there. Based on this, I believe that it is the fault of your application.

SK21, 2025-02-17:
Could I get a copy of the file to test?

SLOWiK83, 2025-02-18:
[Dzialka_.42_2025-01-23_aaa.zip](https://github.com/user-attachments/files/18840488/Dzialka_.42_2025-01-23_aaa.zip)

[image]

SK21, 2025-02-27:
Try beta 8 to see if it works correctly now. The problem was the way gmap.net handles polygons with holes.