SK21/AOG_RC #40: No one fingered way to move the window

mirh, 2025-04-26:
Long story short, there's just no way to move the new 3.8.0+ interface with a touchscreen.

Besides, it's only by pure chance minutes before sending this issue that I figured how to do it with the mouse... I really don't think the right click drag is intuitive (even though I'll grant it's nice to shave off the few pixels of the title bar).

SK21, 2025-04-27:
Long press is equivalent to right-click on a touchscreen. Long press will move the window. I agree it is not intuitive. I am open to suggestions on a better way.

mirh, 2025-04-27:
Uh.. Ok actually that's surprisingly comfy and workable now that I tried it.
And I guess like it's not something that I'd want to remove.

Still I feel like there should be a plan B (I swear I was using 3.9.0 because at least sub-menus could be moved the normal way there).
How about creating a sort of faux title bar: the name of the current product on the left, the X and _ buttons on the right, and all the space in between works the good old dragging way?
(ofc the unit of measurement would have to go away or at least move.. but tbh that seems something that should happen anyway, considering it technically only applies to the first two rows while it's out of place for the quantity and the area when shown separately)

closed 2025-06-17 by SK21